JackBuck/Robo-Plot #6: Create a physical test for motors.

JackBuck, 2016-10-30:
We should be able to use it to:
* Verify the angular accuracy of the motors,
* Verify that the motion of two motor is correctly synchronised.

JackBuck, 2016-11-05:
Printed out protractors and mounted them on card.
Attached a cocktail stick using an elastic band.

I am not sure whether the cocktail stick was moving slightly relative to the rotor (wouldn't be a problem on the small stepper since this has a flat).
I tried blutac as well but that was worse.
Does anyone have any better ideas? Or should we class this as 'good enough'?

JackBuck, 2016-11-05:
@mrmiiike I've added you to the assignees because you were there when we used this!
I was worried about the cocktail stick moving.
Do you have any ideas? (Or ideas re anything else about it!)
If not then we might as well close this issue and leave that extra bit as a `wontfix` for now.

mrmiiike, 2016-11-05:
I think the cocktail stick was the one of the problems, if not, the only problem of why it was slightly off. Till we find a trusted attachment method, I wouldn't worry about it too much for now. Once I get hold of a coupling, then we can test it properly for sure

JackBuck, 2016-11-06:
Yeah, since that's something we'll have fairly soon I think we can close this for now!

closed 2016-11-06 by JackBuck