JackBuck/Robo-Plot #9: Write a process for the team.

Hsenrab, 2016-10-31:
Write a wiki entry outlining the process we are going to follow to (try to!)  keep everything organised. This should be agreed by everyone.

Hsenrab, 2016-10-31:
I am going to write an initial draft, everyone's input and agreement is needed on this. Hopefully because GitHub has a way to raise and close easily we can stick to it better than we did the one we tried last year!

JackBuck, 2016-11-05:
Ok,
Following the meeting on Friday, I'm writing up the final process document now!

JackBuck, 2016-11-05:
So I've done it...
It's quite long and might benefit from being split over a couple of pages in the wiki.
Things which are different to what we decided on Friday are

- The merge commit message convention: I'm not sure you can change the title when merging a pull request through the git hub web interface.
- I've suggested that for complicated issues we merge the master into the issue branch, then test it, then merge the pull request (issue->master). This way we are less likely to break master. Can you think of any objections to this?

Hsenrab, 2016-11-07:
I am happy with this, I suggest you email each team member and ask them to read it and be aware of it as a reference. Once this is done this issue may be closed.

closed 2016-11-07 by JackBuck