JackBuck/Robo-Plot #12: Write and test Motor Class.

Hsenrab, 2016-11-05:
Write a class so we can intuaitively use the motors.

This should include:
Member variables:
GPIO Pins
rps
sequence
on/off bool

Functions:
Set rps
Start()
Start(time)
Stop()
SetSequence
SetDirection()
Step()

These should all be tested.

Hsenrab, 2016-11-09:
This class now sets up the pins and steps them forward according to the sequence. Currently the tests do not stop until the window is manually closed.

JackBuck, 2016-11-16:
Merged in #28 but not yet reviewed

Hsenrab, 2016-11-23:
Unfortunately I had forgotten to add the test functions so this branch has been revived and the tests added. I will look over this and check  I am happy that it is in line with the review changes Jack made before creating a pull request.

JackBuck, 2016-11-25:
@Hsenrab Could you merge the unit tests into M_MotorFunctionality?
(Or alternatively open another pull request)

JackBuck, 2017-01-27:
This was closed with pull request #56 and merged in 6ae8bf6ed46cbc4e45e9e8229055eb66fbfa1f7b

closed 2017-01-27 by JackBuck