JackBuck/Robo-Plot #20: Write functions to generate paths for the shapes we need to draw

JackBuck, 2016-11-05:
What constitutes a path is to be defined by issue #18 .

The shapes we need to draw are
- Straight line
- Circles
- Circular Arcs (the rules document is vague since svg arcs include elliptical arcs, but we will start with circular arcs)

JackBuck, 2017-01-27:
On hold while we investigate converting SVG paths to a series of points.

JackBuck, 2017-02-01:
SVG curves appear to work (superficially!) but there is a bug in the debug_movement.py module which needs fixing before we can be 100% sure!

JackBuck, 2017-02-06:
I just want to test this on an svg file with multiple curves now.

JackBuck, 2017-02-07:
Hopefully just one bug left where the disjoint path making up the bubble man has not been drawn disjoint...

JackBuck, 2017-02-08:
Ok, ready for review!!

Hsenrab, 2017-02-15:
So I added tests note that not all passing as they throw errors. Once these are fixed the outputs for the tests need to generated and committed. @JackBuck

JackBuck, 2017-02-23:
Merged in #78

closed 2017-02-23 by JackBuck