JackBuck/Robo-Plot #23: Decide on Python 2 or Python 3

JackBuck, 2016-11-06:
So when installing openCV on the pi, I re-realised that we haven't decided on python 2 vs python 3 yet.
I've had a quick search and [wiring pi](https://github.com/WiringPi/WiringPi-Python), [openCV3](http://www.pyimagesearch.com/2016/04/18/install-guide-raspberry-pi-3-raspbian-jessie-opencv-3/) and [svgpathtools](https://pypi.python.org/pypi/svgpathtools) all appear to support python 3 (although there is a warning on the svgpathtools page that not everything has been tested in python 3).

The [official python wiki](https://wiki.python.org/moin/Python2orPython3) is definitely trying to convince people to start using python 3. It lists 'mainstream' libraries (such as numpy) which have been ported to python 3. However, they do warn you that porting a library is not a simple task and that you are wise to check before hand that you don't want to use a library which hasn't been ported.

Some of the features have been backported to python 2.7, and some have not. Copying and pasting a (non-exhaustive) list of features which are in python 3 and *not* in python 2:

- strings are Unicode by default
- clean Unicode/bytes separation
- exception chaining
- function annotations
- syntax for keyword-only arguments
- extended tuple unpacking
- non-local variable declarations

We can delay the decision to give us time to decide by using [virtual environments](http://docs.python-guide.org/en/latest/dev/virtualenvs/) to manage versions of dependencies for python 2 / 3 resp. and writing code which can run with both interpreters.
However, I think at some point fairly soon we want to decide which of the two we are going for!

Hsenrab, 2016-11-06:
So I just checked and the emulator is python 3. -I had also forgotten there was a difference. Personally I think we should go with 3. It means we learn the most up-to-date version which makes are skills more transferable. We are unlikely to be too inconvenienced by it and I think if we put the decision off we will end up delaying things with work arounds. Decide now and live with the decision.

JackBuck, 2016-11-06:
Well, having a library which is only python 3 is a big plus for python 3!!
We should keep this open until the **end of the day** in case anyone else has anything to say on it, but I'm keen to not let this decision hold us up!!

JackBuck, 2016-11-06:
Also, for the record, here is a [summary of the main differences](https://www.raspberrypi.org/documentation/usage/python/more.md) between python 2 and python 3.

Hsenrab, 2016-11-06:
Probably worth noting that the emulator is useful but not that much code and I am having to make some tweaks to it anyway.

JackBuck, 2016-11-06:
Ok, keeping this open a little longer - installing openCV for python 3 ( issue #13  ) is appearing a little complicated :S

What appears to be a [very good tutorial](http://www.pyimagesearch.com/2016/04/18/install-guide-raspberry-pi-3-raspbian-jessie-opencv-3/) is possibly more sophisticated than we need. It introduces these virtual environments, which would be fine, except it breaks the wiringpi install (and reinstalling it whilst inside the virtual environment doesn't fix things!).

I've posted a [comment](http://www.pyimagesearch.com/2016/04/18/install-guide-raspberry-pi-3-raspbian-jessie-opencv-3/#comment-410092) to ask his opinion on the matter, and in the mean time I am installing openCV for python2.7.

JackBuck, 2016-11-07:
Ok, I set the build going over night and to my delight discovered that it created binaries for both versions!! So we're all set and didn't need to use the virtual environments :smile:

closed 2016-11-07 by JackBuck