JackBuck/Robo-Plot #27: Take photos with camera.

Hsenrab, 2016-11-10:
Take photos with the camera to help us decide on the height of the camera and how far from the pen it should be.

Photos should be uploaded somewhere and linked here.

Hsenrab, 2016-11-10:
http://raspi.tv/2013/adapt-your-raspberry-pi-camera-for-close-up-use

We may need to adapt the camera to be better at close up if we want to take close up photos in focus

Hsenrab, 2016-11-11:
https://drive.google.com/drive/folders/0B1bLptTMLGBZNEF2aGpPUWNZQzg?usp=sharing

JackBuck, 2016-11-14:
Ooh I forgot about focus! Though looking at the photos you took at 4 & 5 cm, they don't look too bad... The squared paper isn't too clear though!!

Hsenrab, 2016-11-14:
I don't think the fact they are out of focus matters too much. I agree around 4-5cm is the best distance but will have to check if that is a realistic height. Are you happy I close this?

JackBuck, 2016-11-15:
Yeah, close away :smile:

closed 2016-11-23 by Hsenrab