JackBuck/Robo-Plot #34: Acceleration

Hsenrab, 2016-11-18:


Hsenrab, 2016-11-18:
Can you add a description please @LukeAWalker

JackBuck, 2016-11-19:
This is a duplicate of #33 .
Since commits have reference this issue and not the other, I'll close the other and rename this one.

LukeAWalker, 2016-11-21:
Tell me if this is not a description:

Trying to reduce step skipping at high speeds by gradually accelerating to those speeds. In theory stepper motor is less likely to step skip if the momentum of the rotor is already with it.

Hsenrab, 2017-02-23:
I think we have experimented with this and decided that it does not help the skipping of steps. However we need to decide whether we are going to integrate acceleration anyway @LukeAWalker @JackBuck ?

I think that the branch with the acceleration work on should be tagged in case we want to revive the work later but we currently have no intentions of using it.

JackBuck, 2017-03-05:
Given that the result of Luke's investigation was the ramping the speed didn't help, I personally don't think we should merge it. But we should certainly keep it tagged like you say in case we need to return to it!