JackBuck/Robo-Plot #35: Create a neat API for driving the servo motor

JackBuck, 2016-11-19:
The working script from #3 is below.
Assuming I noted it down correctly after the session, the working range for the input parameter is from 0.028 to 1.3.

```python
# Working range is from 0.028 to 1.3

# Note: Command line syntax for this module
# Servo.py 0.5
# Servo.py 0.1
#
# Takes 1 argument which is floating number 0<x<1

# import wiringpi python wrapper
import wiringpi
import sys
import time

# Raspberry PI BCM 18 = only PWM pin (according to the internet)
SERVO = 18
PWM_RANGE = 500
PWM_DIVISOR = 765

# Initialise PWM pin
wiringpi.wiringPiSetupGpio()  # Setup wiringpi
wiringpi.pinMode(SERVO, wiringpi.PWM_OUTPUT)  # Set SERVO pin as PWM output
wiringpi.pwmWrite(SERVO, 0)  # Turn output off

# Setup PWM
wiringpi.pwmSetMode(wiringpi.PWM_MODE_MS)  # Set PWM mode as mark space (as opposed to balanced - the default)
wiringpi.pwmSetRange(PWM_RANGE)  # Set PWM range (range of duty cycles)
wiringpi.pwmSetClock(PWM_DIVISOR)  # Set PWM clock divisor
# Note: PWM Frequency = 19.2MHz / (PWM_DIVISOR * PWM_RANGE)

# Set output as defined by command line
PWM_OUTPUT = int(float(sys.argv[1]) * PWM_RANGE)
print(PWM_OUTPUT)

# If command line input > 1 then set to maximum
# if command line inptu < 0 then set to minimum
if PWM_OUTPUT > PWM_RANGE:
    PWM_OUTPUT = PWM_RANGE
elif PWM_OUTPUT < 0:
    PWM_OUTPUT = 0

wiringpi.pwmWrite(SERVO, PWM_OUTPUT)

time.sleep(1)
```

JackBuck, 2016-11-19:
@LukeAWalker  does this working range sound right? I've noticed your comment which says the input should be <1. Could I have mis-wrote it down instead of 0.13?
Since you have the kit, next time you've got it set up could you verify that for us? Thanks!!

LukeAWalker, 2016-11-21:
0 - 1 is the minimum and maximum for development mode (i.e. less than 0 and more than 1 isn't physically possible to obtain). From experimentation, out of that range the useful range is 0.028 to 0.13. 

For ease of use, i'd say that the working range was 0.03 to 0.13, then you have a working range of 0.1

JackBuck, 2017-02-21:
All that remains for this issue is to
* fake the wiringpi module
* test the code
* set the 'up' and 'down' pen positions at a practical session

JackBuck, 2017-03-18:
This has now been merged

closed 2017-03-18 by JackBuck