JackBuck/Robo-Plot #38: Define pin/GPIO allocations for motors, LEDs, encoders, etc.

LukeAWalker, 2016-11-21:
Define permanent allocations for the GPIO pins on the raspberry pi.

JackBuck, 2016-11-22:
I believe that we cannot set-up GPIO pins more than once (though this may simply be the emulator warning us by raising an exception).

Do we think there is a case for some sort of `HardwareRepository` static class or module to manage handing out a singleton instance of each piece of hardware? It would also be the one piece of code responsible for hard coding the GPIO pins (hence why I'm commenting here!).

@Hsenrab

Hsenrab, 2016-11-23:
I think that would work nicely - we cant then just call a SetupGPIOpins() function at the start of our program and know everything will be set up from then on.

JackBuck, 2017-01-09:
I believe this is all done now.

closed 2017-01-09 by JackBuck