JackBuck/Robo-Plot #39: Define a project structure

JackBuck, 2016-11-22:
The number of 'loose' files we have is starting to become unmanageable and we need a 'synchronised' package / module structure to be adopted in all branches.

I was thinking something like:

- Robo-Plot
  - `__init__.py`
  - resources
    - sample_path_images_1
    - *Here we would keep any resources we need for the scripts to run. The folder would be in the .gitignore, but by placing it in the project structure, we can use **relative paths** to reference files from the code. This way, we don't need to change filepaths before we can run each others code. Those running linux retain the freedom of being able to keep their resources wherever they like and simply symlink them here (which can be useful for say keeping lots of image files from clogging up a backup process).*
  - scripts
    - *Here we would put all the scripts we want to run during the lab sessions. These would typically take commandline arguments to make for easy testing.*
  - core
    - `__init__.py`
    - `gpio_connections.py`
    - `test_gpio_connections.py`
    - *any other shared modules*
  - imgproc
    - `__init__.py`
    - *image processing modules here*
  - svg
    - `__init__.py`
    - *reading and converting `.svg` files to instructions for the core package here*

Further, for consistency with module / package naming:

- As per [PEP8](https://www.python.org/dev/peps/pep-0008/#package-and-module-names):
  > Modules should have short, all-lowercase names. Underscores can be used in the module name if it improves readability. Python packages should also have short, all-lowercase names, although the use of underscores is discouraged. 
- Unit test modules should be named `test<module-name>` or `test_<module-name>` as appropriate. This fits in with the unittest module's default [discovery](https://docs.python.org/3.4/library/unittest.html#unittest-test-discovery) pattern.

Hsenrab, 2016-11-23:
In order for this to work effectively we need to make this change merge it to the master and then have the change merged to all active branches. Otherwise when people add files etc they will be added in the wrong place in this new structure and the automatic git merges will not realise this. We should then as part of the review for each branch check it is all in this new format before merging it back to the master.

JackBuck, 2016-11-23:
The auto merging thing is very true, and merging to master does seem like the simplest way... Though merging into master means we need to finish each milestone branch which we 'd want to merge in!
(It would also pay to remember at this point that since git tracks files not folders, we cannot simply create say the empty `core` folder on Iss#28_BasicLineIdentification. So we are always going to need some degree of head-screwed-on-ness when creating folders one one branch which are new to another.)

JackBuck, 2016-11-25:
Yeah, that sounds like a good idea!
Also, I was thinking that this would be a good point to share (resp. ignore) the files we (don't) want from the .idea directory. I found [this example](https://github.com/github/gitignore/blob/master/Global/JetBrains.gitignore) of which files we might want to ignore.

JackBuck, 2017-01-08:
I've updated my initial comment to include a `roboplot` directory (so that we can actually import the top-level package: it turns out that you are not allowed hyphens in python module names)

JackBuck, 2017-01-08:
@Hsenrab To save repeated documentation, I suggest we drop the readme file in actual python packages and use the docstring in the `__init__.py` file instead. For what it's worth, then we can say
    `import roboplot`
    `help(roboplot)`
in a python console, and the docstring will be printed as part of the output.

JackBuck, 2017-01-08:
Ok, I've created a separate issue #52 for the `.idea` sharing.

JackBuck, 2017-01-09:
This is now merged into `master` and the `M_MotorFunctionality` branch.
Please follow the new structure when merging into either of these.
It would also be prudent to update any branches you are working on asap, by merging the changes up the branch levels (i.e. `master`->*milestone*->*issue*).

JackBuck, 2017-01-27:
Closing this issue now, since the new structure is in all the main branches (though not all branches!).

closed 2017-01-27 by JackBuck