JackBuck/Robo-Plot #69: Fix point discrepencies in point order passed to debug_movement.py

JackBuck, 2017-02-01:
- Currently in all versions the `AxisPair.current_location` property is passed to `DebugImage.add_point(...)`.
- The `AxisPair.current_location` property is a point with coordinates (x,y).
- In all versions, the `DebugImage.addpoint(...)` method is expecting a point ordered (y,x).

Hsenrab, 2017-02-06:
Well spotted! Not sure how its working at all ... will look at this tonight.

Hsenrab, 2017-02-06:
Think the way you have separated out the debugging is good much cleaner than what I had!
Need to:
Document add_point to make the x,y ordering clear.
Decide whether we are going with (y,x) or (x,y) 
Determine if it is currently working and if so why.
Fix it.

JackBuck, 2017-02-06:
@Hsenrab Since I need this now and can't get hold of you, I'm assuming you're not currently working on this!
I'll do it now 😄

JackBuck, 2017-02-06:
Change now merged into master, and propagated into #20 (where the svg curve is being developed ).

closed 2017-02-06 by JackBuck

Hsenrab, 2017-02-07:
Reviewed changes look fine runs as expected.